TLeconte/vdlm2dec #1: "decode more vdlm2 protocols (looking for info)"

DickvanNoort, 2017-08-10:
Hi Thierry,
Not sure how to contact you directly, so I just created an "issue" ;-) 
You mention "decode more vdlm2 protocols (looking for info)"
Have you seen the dumpvdl2 project of Tomasz?
He has implemented a lot of messages as well as multiple freq decoding.
I have been running it for several months now and also went to EHAM to catch upload and broadcast messages.

Cheers,
Dick

TLeconte, 2017-08-19:
Hi
Yes Github lacks a forum function (or I don't known how to activate it)

No, I did not known dumpvdl2, I will look at it and perhaps learn some interesting things.
I don't really want to do a protocol analyser  and in particular not display all this iso layer craziness. My first goal was to decode acars and made vdl2dec a successor of my acarsdec program. I add  some XID decoding for interesting info, like rough position and destination airport. I could add more but it is essentially protocol info not really interesting.

I work on multifrequency decoding and vdl2dev will use the same front end than acarsdec.

DickvanNoort, 2017-08-19:
Hi,

Yes, you will see that dumpvdl2 tries to decode it all, including all protocol stuff and CPDLC messages.

Well if you have something ready to test/play just drop me a PM line at hobbyd@nowo.nl.

Cheers,
Dick   

From: Thierry Leconte 
Sent: 19 August, 2017 18:45
To: TLeconte/vdlm2dec 
Cc: DickvanNoort ; Author 
Subject: Re: [TLeconte/vdlm2dec] "decode more vdlm2 protocols (looking for info)" (#1)

Hi
Yes Github lacks a forum function (or I don't known how to activate it)

No, I did not known dumpvdl2, I will look at it and perhaps learn some interesting things.
I don't really want to do a protocol analyser and in particular not display all this iso layer craziness. My first goal was to decode acars and made vdl2dec a successor of my acarsdec program. I add some XID decoding for interesting info, like rough position and destination airport. I could add more but it is essentially protocol info not really interesting.

I work on multifrequency decoding and vdl2dev will use the same front end than acarsdec.

TLeconte, 2018-01-25:
private xid c0 to c8  (ground station information) decoding has been added.
I don't plan to go further as I am mostly interested in ACARS. (works are in progress to add  better inter-operation with acarsdec/acarsserv).

closed 2018-01-25 by TLeconte