TLeconte/vdlm2dec #4: VDL2 Ground Station Broadcast Location

acarslogger, 2018-01-20:
A Broadcast Message transmitted from a VDL2 ground station usually contains a local airport name and location e.g a 4 character ICAO code (EGLC) and a latitude and longitude location.
Can vdlm2dec also decode this data and add it as minimum to the "-v 1" normal output logs?

A Broadcast message in vdlm2dec -v 3 format from vdl2 ground station 260F37 EGLC.

```
ppm -0.2
#5 19/01/2018 11:03:44.292
Command from Ground: 260f37 to: ffffff 
Frame-U: -/- XID
Parameter set ID
Procedure classes
HDLC options
Specific options: VDA:ESN:IHN:BHN:BCN
unknown private id c1
c1 04 45 47 4c 43                                  |..EGLC..........|
unknown private id c4
c4 06 53 49 54 00 00 01                            |..SIT...........|
unknown private id c5
c5 04 94 00 00 00                                  |................|
unknown private id c8
c8 03 20 30 00                                     |.. 0............|
-----------------------------------------------------------------------
```
The definition is in RTCA DO 224C.

The 4 character ICAO airport code EGLC is fairly obvious!

The latitude and longitude are encoded in the last 3 bytes or your message above e.g. 20 30 00 
12 bits for the latitude, 12 bits for longitude.
First bit is sign; 0 = positive = North or East.
If sign = 1 = negative = South or West. Then need twos-compliment.
Remaining 11 bits are the value.
For EGLC latitude value is 515 = 51.5N, longitude = 0 = 0.0E (maybe not the best example to use!)

acarslogger, 2018-01-20:
A better example of the latitude and longitude decoding.
20 DF EE
Translates to 52.5N 1.8W (Birmingham EGBB)

TLeconte, 2018-01-21:
I add private ground station xid decoding  (C0 to C8), but, has I said, I don't receive ground stations so I can't test.
Could you test my last commit [](https://github.com/TLeconte/vdlm2dec/commit/25df8428bb7756d9791550fec0068b3ea37abe0c) , and send me some logs wit -v 3

TLeconte, 2018-01-21:
Edit : hummm, negative long are probably wrong. Will see that lateer

acarslogger, 2018-01-21:
Not sure if it's you or me, but I've updated my Raspberry Pi with the latest version (includes changes from the last 24 hours), and now I'm not getting any output in the log file! I don't think I've changed anything at my end but I will investigate further.

TLeconte, 2018-01-22:
I fixed lat/lon .
I would appreciate test wit ground station reception.

acarslogger, 2018-01-22:
I'll test it when I can get vdlm2dec working again. As I said above I'm not getting anything written to the log file since you've incorporated the recent changes. I'm using an RTL-SDR v3. I'll continue to investigate.
Is this the same issue as issue 6 where a clean install of vdlm2dec  **Does not result in any decodes..**

acarslogger, 2018-01-24:
Actual Broadcast Message received using the latest version (still v2.0?) of vdlm2dec.

> **[#5 (F:136.975 P:-2.0) 23/01/2018 22:23:24.665 --------------------------------
> Command from Ground: 262257 to FFFFFF 
> Frame-U: XID
> Specific options: VDA:ESN:IHN:BHN:BCN
> Frequency support : 136.870000 (2) 262241 136.870000 (2) 262243 136.870000 (2) 262242 
> Airport coverage : EGLL 
> ATN router nets : ADM: 534954 ARS : 000001
> Station system mask : 200000
> Station Position 51.5  -0.4**

The Airport Coverage ICAO code and the Station Position are correct.
The Frequency support ground station ids are also correct, although I think the frequency data in the broadcast message is only 2 inferred decimal places.
**From DO-224C: As an example, for a frequency of 131.725 MHz, the encoded value is decimal 3172 or hexadecimal C64.**

So maybe either display 136.87 or use the airband frequency of 136.875.
I assume the (2) is the modulation - D8PSK, Mode 2, 31500 bps
Thanks.

TLeconte, 2018-01-24:
ok
I fixed freq display by displaying only  2 decimals. (what a strange specification ...)

closed 2018-01-24 by TLeconte