TLeconte/vdlm2dec #7: Feature: Use RTL serialnumber to select the dongle

DickvanNoort, 2018-01-20:
When one has multiple dongles connected using the device ID can be confusing.
Please implement the use of serialnumber.
There are several examples on github that have this feature (like dumpvdl2).. 

Cheers,
Dick

DickvanNoort, 2018-01-21:
Appologies, I was interpreting the error messages wrongly.
The program already support using the SN of the dongle.

Cheers,
Dick

closed 2018-01-21 by DickvanNoort

TLeconte, 2018-01-21:
yes but it's nor really clear in the doc.
I have to do something about it.

DickvanNoort, 2018-01-22:
Hi Thierry,

As long as the program is not stable and only a few testers are suing it the documenting the features and arguments could be a low priority.

The current version which I downloaded and compiled this morning does not decode at all..
But I suppose you already know that.

Cheers,
Dick

From: Thierry Leconte 
Sent: Sunday, January 21, 2018 21:21
To: TLeconte/vdlm2dec 
Cc: DickvanNoort ; State change 
Subject: Re: [TLeconte/vdlm2dec] Feature: Use RTL serialnumber to select the dongle (#7)

yes but it's nor really clear in the doc.
I have to do something about it.