TLeconte/vdlm2dec #10: Specific sequence of arguments cause warnings

DickvanNoort, 2018-01-22:
When I start vdlm2dec this way:
/home/pi/vdlm2dec/vdlm2dec -v 2 -r 11 -p -1  136.725 136.775 136.825 136.875 136.975 -l "/home/pi/vdlm2dec/logs/vdl2_rtlsdr-11.log" >> "/home/pi/vdlm2dec/logs/process-vdlm2dec.log" 2>&1 &  

The program starts and works as expected but gives 4 warnings:
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000011

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain : 49.600000
WARNING: Invalid frequency 0
WARNING: Invalid frequency -1000000
WARNING: Invalid frequency 0
WARNING: Invalid frequency 0
Set center freq. to 137025000Hz
Exact sample rate is: 2000000.052982 Hz

2 of the warnings go away when I remove the "-p -1" part like this:
/home/pi/vdlm2dec/vdlm2dec -v 2 -r 11  136.725 136.775 136.825 136.875 136.975 -l "/home/pi/vdlm2dec/logs/vdl2_rtlsdr-11.log" >> "/home/pi/vdlm2dec/logs/process-vdlm2dec.log" 2>&1 & 

All warnings go away if the arguments are move to another place like this:
/home/pi/vdlm2dec/vdlm2dec -v 2 -l "/home/pi/vdlm2dec/logs/vdl2_rtlsdr-11.log" -p -1 -r 00000011 136.725 136.775 136.825 136.875 136.975 >> "/home/pi/vdlm2dec/logs/process-vdlm2dec.log" 2>&1 &

Cheers,
Dick

TLeconte, 2018-01-22:
Yes, it's not clear in the doc, but frequencies must be last. ie  : 

/home/pi/vdlm2dec/vdlm2dec -v 2 -r 11 -p -1 -l "/home/pi/vdlm2dec/logs/vdl2_rtlsdr-11.log" 136.725 136.775 136.825 136.875 136.975  >> "/home/pi/vdlm2dec/logs/process-vdlm2dec.log" 2>&1 &

Note : do you really have eleven rtl dongles ?

closed 2018-01-22 by TLeconte

DickvanNoort, 2018-01-22:
> Note : do you really have eleven rtl dongles ?
No only these 10:

# Overview dongles and their freq correction.
# ID 1    sn 00000001    (+54)    "white light"
# ID 2    sn 00000001    (??)    "white light"
# ID 3    sn 00000003    (+50)    "white light"
# ID 4    sn 00000004    (+70)    "white light"
# ID 5    sn 00000005    (+70)    "blue light"
# ID 7     sn 00000007    (+73)    "blue light"
# ID 10    sn 00000010    (+0)    "R820T2 metal case"  
# ID 11    sn 00000011    (-1)    "R820T2 metal case" 
# ID 12    sn 00000012    (-2)    "R820T2 metal case"
# ID 13    sn 00000013    (-1)    "R820T2 metal case"

Cheers,
Dick