TLeconte/vdlm2dec #9: Current version not decoding.

DickvanNoort, 2018-01-22:
Hi,
The current version which I downloaded and compiled this morning does not decode at all..
The most simple variant 
./vdlm2dec -r 11 136.975

Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000011

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain : 49.600000
Set center freq. to 137025000Hz
Exact sample rate is: 2000000.052982 Hz

Does not result in any decodes..  
I suppose you already know that, but just to be sure.....

Maybe you can start using the 'Master' and 'Unsafe' feature in Github. That way one can always easily fallback to the Master (working) version.

TLeconte, 2018-01-22:
No I don't know that
exact syntax is : 
./vdlm2dec -r 0 136.975
but it must not be the problem.

Yes, I could use branches, but in general, I only commit tested version. The problem is that I don't have a rtl dongle any more (they are not so reliable) and recently, I test only with my airspy ...
I have to buy another one to test.
Current HEAD version is working with my airspy / tegra , but it's true that I have to test with less exotic configs.

TLeconte, 2018-01-22:
Ok something seems broken with rtl.
I just bought a new rtl sdr,  will test and fix soon.
sorry

DickvanNoort, 2018-01-22:
>  No I don't know that
>  exact syntax is 
>  ./vdlm2dec -r 0 136.975
>  but it must not be the problem.
The syntax with the -r 11 and -r 00000011 works well.  
Sorry, just assumed you had an RTL dongle.
Well you know “assumption is the mother of all ....”.  

So it should be a very recent change that broke it..

Good luck.

TLeconte, 2018-01-23:
I had a rtl dongle , but they are not really reliable.
I have an otherone in command, will get it end of week.
In the meantime, I just commit a blind change to rtl.c , fixing an obvious error.
could you try it ?

DickvanNoort, 2018-01-23:
Good blind change!!
This version works fine..
Thank you.

TLeconte, 2018-01-23:
great !

closed 2018-01-23 by TLeconte