TLeconte/vdlm2dec #12: Id Status (Request)

acarslogger, 2018-01-29:
Would it be possible to enhance vdlm2dec messages to include whether an id is an aircraft (and it's status) or a ground station? I think this information is avilable to you when decoding the VDL2 message.
As an example the current message format shows
Command from : 406CD0 (on ground) to : 10919A

Could this be changed to show something similar to
Command from : 406CD0 (aircraft, on ground) to : 10919A (ground station)
or
Command from : 406CD0 (aircraft, airborne) to : 10919A (ground station)
or
Command from : 10919A (ground station) to : 406CD0 (aircraft, on ground)
or
Command from : 10919A (ground station) to : 406CD0 (aircraft, airborne)

Thanks
Stuart

TLeconte, 2018-01-29:
done  : 

> [#3 (F:136.975 P:-5.4) 29/01/2018 21:45:41.634 --------------------------------
Command from : Aircraft:406E69 (airborne) to : GroundD:10925A 
Frame-I: Ns:6 Nr:4

closed 2018-01-29 by TLeconte

acarslogger, 2018-01-30:
Thanks.
I'll test it when I return home.
Cosmetic/Typo comment - You show "GroundD:10925A". Should Ground**D** be Ground Id or Ground Station?
Stuart

TLeconte, 2018-01-30:
Not really a typo. 
In fact in vdlm2 addresses are on 27bits.
The first 3 bits gives the address type : 

 0 reserved   Future   use            
 1 Aircraft  24-bit ICAO address  
 2  reserved   Future  use            
 3  reserved  Future  use            
 4 Ground station  ICAO-administered address space 
 5 Ground station ICAO-delegated address space 
 6  reserved  Future  use            
 7 All stations broadcast 

So GroundD is type 5 ( **D**elegated address space)
and GroundA is type 4 ( **A**dministered address space  )

acarslogger, 2018-02-01:
Thanks for information. I'm getting GroundD messages on uplinks and downlinks. I've haven't seen a GroundA.