TLeconte/vdlm2dec #14: Odd status "T0"  (T zero).

DickvanNoort, 2018-02-04:
Hi Thierry,
Found a couple of these messages in the log using version V2.0 (ab8896d) 
Observe the  "T0" between "to" and ":0A0010"..

[#5 (F:136.975 P:-3.9) 02/02/2018 11:02:16.690 --------------------------------
Response from Aircraft:AA6C50 (airborne) to T0:0A0010
Frame-U: DM

Dumpvdl2 shows them with a status "Reserved". 
[2018-02-02 11:02:16 GMT] [136.975] [-42.1/-51.8 dBFS] [9.6 dB]
AA6C50 (Aircraft, Airborne) -> 0A0010 (reserved): Response
AVLC: type: U (DM) P/F: 0

Cheers,
Dick

TLeconte, 2018-02-05:
As I have said in #12 , in vdlm2 addresses have a type . There are 8 types but only 3 are defined (type 1 aircraft, type 4 and 5 ground stations) . All the others types are "reserved for future use".
T0 means type 0 address witch is surprising , but the address is surprisingly simple too...
Perhaps a station in test or not correctly setted ?
Do you receive packet from the station ?

DickvanNoort, 2018-02-05:
Sorry for not relating the response in #12 to this. 
At first I thought is was a bug, later when more of those messages appeared (and I could also find them in my dumpvdl2 log) I understood something else was going on.
Earlier we (Stuart and I) discussed this with Tomasz (the author of dumpvdl2) but did not come to a conclusion. Yes it is documented as "reserved" but it seems to be used for yet unknown reasons. 
Yesterdays log (2018-02-04) contained 277 of those messages. 
I still have to analyse them in more detail but so far I can tell that:
a. All but two are from the same aircraft (A3C4D2) and T0 as well as T3 values were used.
First message was received at 16:42:35.328 the last at 16:48:36.498. 
The "from" ICAO belongs to a N342AX. All "to" ICAO's seem invalid for real aircraft.
b. The other two are at a total different time:
09:47:13.432  Response from T3:3409D4 (airborne) to GroundD:10980B
19:29:29.163  Response from T6:004440 (airborne) to T6:7A2211
c. During this same period there where 106 "Aircraft to Aircraft" messages.
All are from the same aircraft (A3C4D2) and to the same aircraft (8B060D) which is not an allocated ICAO24-address. 

Based on internet sources N342AX was in my area at that time but I did not find it in my ADS-B log which as a limited reach and unfortunately was not in sharing mode.. 

Attached is the logfile covering that time period.

Cheers,
Dick
[vdl2_rtlsdr-11_20180204 (selection).log](https://github.com/TLeconte/vdlm2dec/files/1696719/vdl2_rtlsdr-11_20180204.selection.log)

TLeconte, 2018-02-06:
Seeing the log I guess a broken vdlm2 management device on  A3C4D2.
It sends tons of packets to random , malformed addresses ...

closed 2018-02-06 by TLeconte