TLeconte/vdlm2dec #20: JSON Output - Enhancement

acarslogger, 2018-02-09:
Hi Thierry,
As well as the normal acars data, you have included an additional packet in the JSON outputs (-j and -J) containing the destination. For completeness would it be possible to enhance the JSON outputs (-j and -J) to include all aviation related data for all messages e.g. Aircraft ICAO, Ground Station Id, Direction/Uplink/Downlink indicator, Location (Lat & Lon), altitude and destination.

Examples.

> [#5 (F:136.975 P:-6.6) 08/02/2018 00:00:13.620 --------------------------------
> Response from Aircraft:987654 (airborne) to GroundD:214A77 
> Frame-S: Nr:1 RR

JSON output would include Date, Time, Channel, Frequency, Aircraft ICAO, Ground Station Id, Direction/Uplink/Downlink indicator.

> [#5 (F:136.975 P:-4.7) 08/02/2018 00:01:40.879 --------------------------------
> Command from Aircraft:3C66B8 (airborne) to GroundD:10980A 
> Frame-U: XID
> Connection management: HO|VDA|ESN
> XID sequencing 0:3
> Specific options: GDA:ESN:IHS:BHN:BCN
> Destination airport EGCC
> Aircraft Position 51.3   4.0 alt: 36000

JSON output would include Date, Time, Channel, Frequency, Aircraft ICAO, Ground Station Id, Direction/Uplink/Downlink indicator, Latitude, Longitude, Altitude and Destination.

Thanks
Stuart

TLeconte, 2018-02-10:
done  : adding lat/lon/alt 
Not sure I will add all the other informations in json.

acarslogger, 2018-02-11:
Thanks for the lat, lon and altitude.

In your normal log file every VDL2 message starts with

> [#5 (F:136.975 P:-6.6) 08/02/2018 00:00:13.620 --------------------------------
> Response from Aircraft:987654 (airborne) to GroundD:214A77

So for completeness I think it would be beneficial to include this data in the JSON outputs.
Every VDL2 message would generate a JSON output with Frequency, Date/Time, Aircraft ICAO, Ground Station Id and a Direction/Uplink/Downlink indicator.
For VDL2 messages that have additional data (i.e. a message with acars, or a messages with lat, lon alt, destination), this could either be appended to the basic JSON output above, or created as separate JSON outputs.
That's my thoughts. It's your application, so you must do whatever is best for the functionality and output that you want the application to provide.
Thanks for your efforts in creating and enhancing this application.
Stuart

TLeconte, 2018-09-22:
Just add -U -E and -G options for more configurable  json output

Hopes it's sufficient

closed 2018-09-22 by TLeconte