TLeconte/vdlm2dec #22: JSON output - duplicated data and bad format

acarslogger, 2018-02-14:
Hi Thierry,

I've found some JSON format error / duplicated acars output in the JSON output.
There are 2 examples below. I've included them in this 1 Github issue as I suspect that are all closely linked to one area of the source code.

The examples include the default log file output and the equivalent JSON output.

**Example 1**

BA018F (Msg No M33A) sent an FST01 message.
There were then 2 acars messages without an actual acars message.
Then BA13GV (Msg. no: D00A) sent a long H1 acars message.

```
[#4 (F:136.875 P:1.8) 14/02/2018 15:14:47.266 --------------------------------
Command from Aircraft:400D89 (airborne) to GroundD:262243 
Frame-I: Ns:2 Nr:1
ACARS
Aircraft reg: G-MEDL Flight id: BA018F
Mode: 2 Msg. label: 15
Block id: 9 Ack: !
Msg. no: M33A
Message :
FST01EGLLEGPDN052435W0007268236  94  16- 3C 61273      388  16071514

[#5 (F:136.975 P:-2.4) 14/02/2018 15:14:47.282 --------------------------------
Command from Aircraft:3C5EF4 (airborne) to GroundD:10983A 
Frame-I: Ns:5 Nr:5

[#5 (F:136.975 P:-2.1) 14/02/2018 15:14:47.788 --------------------------------
Command from Aircraft:461F30 (airborne) to GroundD:214AB7 
Frame-I: Ns:6 Nr:7

[#5 (F:136.975 P:-4.7) 14/02/2018 15:14:48.314 --------------------------------
Response from Aircraft:406230 (airborne) to GroundD:260F37 
Frame-U: DM

[#4 (F:136.875 P:1.9) 14/02/2018 15:14:48.577 --------------------------------
Response from Aircraft:400D89 (airborne) to GroundD:262243 
Frame-S: Nr:2 RR

[#5 (F:136.975 P:-2.0) 14/02/2018 15:14:49.101 --------------------------------
Response from Aircraft:461F30 (airborne) to GroundD:214AB7 
Frame-S: Nr:0 RR

[#5 (F:136.975 P:-0.4) 14/02/2018 15:14:50.165 --------------------------------
Command from Aircraft:4074B0 (airborne) to GroundD:10902A 
Frame-I: Ns:0 Nr:3
ACARS
Aircraft reg: G-EZRR Flight id: U289DM
Mode: 2 Msg. label: _d
Block id: 4 Ack: O
Msg. no: S87A
Message :

[#2 (F:136.775 P:-3.6) 14/02/2018 15:14:50.819 --------------------------------
Response from Aircraft:4BAA6B (airborne) to GroundD:275D25 
Frame-S: Nr:2 RR

[#5 (F:136.975 P:-4.2) 14/02/2018 15:14:50.836 --------------------------------
Response from Aircraft:3C5EF5 (airborne) to GroundD:10914A 
Frame-S: Nr:5 SREJ

[#5 (F:136.975 P:-2.2) 14/02/2018 15:14:50.885 --------------------------------
Response from Aircraft:3C5EF4 (airborne) to GroundD:10983A 
Frame-S: Nr:6 RR

[#5 (F:136.975 P:-2.2) 14/02/2018 15:14:51.065 --------------------------------
Command from Aircraft:3991E9 (airborne) to GroundD:10919A 
Frame-I: Ns:6 Nr:7
[... 130 lines cut ...]
Frame-I: Ns:3 Nr:4
ACARS
Aircraft reg: HA-LXR Flight id: W609AY
Mode: 2 Msg. label: H1
Block id: 2 Ack: !
Msg. no: D02A
Message :
#DFBA321,007163,1,1,TB000000/REP048,00,00,1/HA-LXR,14FEB18,144905,EGGW,EPKT,W60009/TIA05WIZ090501,01,  5.00,  6.25/HA14133718,134309,14134341,134638/HB14143142,143320,14143454,14144846,1/AA134139,0,134703,0/AB141007,0,144117,0/AC000000,0,000000,0/AD000000,0,000000,0/KA134143,0,134700,0/KB141015,0,144117,0/KC000000,0,000000,0/KD000000,0,000000,0/BA141020,144018,  8.38,  4.75/BB000000,000000,  0.00,  0.00/BC000000,000000,  0.00,  0.00/BD000000,000000,  0.00,  0.00/CA134653,0,142718,0/CB000000,0,000000,0/CC000000,0,000000,0/CD000000,0,000000,0/DA134700,0,141015,0/DB000000,0,000000,0/DC000000,0,000000,0/DD000000,0,000000,0/EA000/EB000/EC000/ED035/EE035/EF029/EG023/EH040/FA01146,03157/FB01146,03159/FC07099/GA0139/MA  527/NA  51.8804, 0.000168/PA-  0.3746, 0.000092/QA035,000/RA 19.75, 19.00, 19.50, 18.75/SA1433,1469/TA135018,0,141153/UA135022,0,141146,  8440/:

```

The equivalent JSON output shows

```
{"timestamp":1518619751.496139,"station_id":"Pi3","channel":0,"freq":136.725,"icao":12592061,"mode":"2","label":"H1","block_id":"6","ack":"!","tail":"C-FNNW","flight":"AC0855","msgno":"D74M","text":"#DFB37 13 12 12 131334\r\n002211  0  01001000045474532 13 12 12 131332\r\n002211  0  01001000045314529 13 12 12 131331"}
{"timestamp":1518619753.1174209,"station_id":"Pi3","channel":4,"freq":136.975,"icao":10807330,"mode":"2","label":"_d","block_id":"2","ack":"K","tail":"N415QS","flight":"GS415Q","msgno":"M60A"}
{"timestamp":1518619755.657136,"station_id":"Pi3","channel":4,"freq":136.975,"icao":4661107,"mode":"2","label":"H1","block_id":"2","ack":"!","tail":"HA-LXR","flight":"W609AY","msgno":"D02A","text"::"#DFB37 13 12 12 131334\r\n002211  0  01001000045474532 13 12 12 131332\r\n002211  0  01001000045314529 13 12 12 131331"}
```

The W609AY H1 message has been replaced with the original AC0855 '#DFB37 13 12 12 1313...' message and the json format has been corrupted to show "text"::"#DFB37 13 12 1 . It's now got 2 colons after the text token.

I can supply more examples if required.
I haven't checked yet if this also applies to acarsdec.
Regards
Stuart

TLeconte, 2018-03-11:
really strange.
It's could only be a problem with the cJSON lib or a bad use of the lib by vdlm2dec, because it outputs a just plain  bad json
Alas, its contributed code that I did not write, so I have to try to understand how the json lib works ...

TLeconte, 2018-03-11:
could you check commit 66e376fecc6c316ce8da45893c30a7370abb14c1

acarslogger, 2018-03-12:
I have the new version up and running. Looks OK at the moment, but need to wait until I have a larger log file to analyse properly.
Stuart

acarslogger, 2018-03-14:
Hi Thierry.
I've looked through the logs from the last 24 hours and everything looks good. I couldn't find any duplicates or corrupt data.
Thanks

TLeconte, 2018-03-14:
Great.
By the way acarsdec had the same problem 
Thanks fot testing

closed 2018-03-14 by TLeconte