TLeconte/vdlm2dec #26: More lines of unknown datas

ziorick, 2018-09-08:
[#1 (F:136.975 P:-20.2) 08/09/2018 22:36:08.327 --------------------------------
Command from Aircraft:344583 (airborne) to GroundD:10F65A 
Frame-I: Ns:3 Nr:6
unknown data

or

[#1 (F:136.975 P:-21.7) 08/09/2018 22:35:53.483 --------------------------------
Response from GroundD:10F65A (on ground) to Aircraft:3C5EED 
Frame-S: Nr:4 RR

I reiceive more and more lines like these above. 
Can I have the choice to store the data (or a track of it) in the database's acarsserv? I use your software and it's great!

TLeconte, 2018-09-09:
Not so easy.
Acarsserv stores acars messages in a database, but vdl mode 2 transmit not only acars but other data somes are stream and not messages oriented. Not easy to store that in a database designed for acars messages.
And for compatibility reasons I don't want to change the database scheme.

What could be done , is to add these frames to the json output. There are already some "not acars" frames  that are decoded and send in json . Ex :  
{"timestamp":1536479723.4279661,"station_id":"radioser","channel":1,"freq":136.775,"icao":4197252,"dsta":"LPFR","lat":47.700000762939453,"lon":-3.7000000476837158,"alt":34000}

I could add the not decoded frame to json, but I don't really see the usage. By the way, your 2nd example of frame, is just a frame without any data (protocol frame). The only reason to log it is to record the plane addr.

In fact acarsserv was done for acarsdec originaly.  I use it for vdlm2dec acars too, but a new json oriented dababase will be the real solution for full vdlm2 recording. (and thare will always the pb of data that are strams and not messages).

Perhap's I will add an option to output all the received frames to json, but I will not change acarsserv.

ziorick, 2018-09-10:
There are some messages (appers to be ACARS) that can't be stored or decoded like that:
[#1 (F:136.975 P:-22.2) 10/09/2018 10:09:52.422 --------------------------------
Command from Aircraft:4CA8F7 (airborne) to GroundD:212437 
Frame-I: Ns:3 Nr:2
ACARS
Aircraft reg: EI-IMJ Flight id: AZ0343
Mode: 2 Msg. label: H1
Block id: 1 Ack: !
Msg. no: D16A
Message :
#DFBA01/A31901,1,1/CCEI-IMJ,SEP10,095330,LFMN,LIRF,0343/C106,92704,5000,46,0010,0,0100,46,X/CEN152,29015,250,655,5490,296,C81011/EC729802,00322,00387,00271,73,09/EE779679,36155,00371,25268,73/N10769,0769,0906,5336,1053,XXXXX/N20768,0769,0911,5495,1071,XXXXX/S1XXXXX,0588,1174,3888,XXXX,048,048/S2XXXXX,0586,1176,3991,XXXX,039,033/T1100,086,026,36,089,06114,0255/T2100,086,017,39,068,06149,0253/V110,06,068,169,01,00,00000/V206,10,221,312,01,00,00000/V3XX,XX,XXX,XXX,XXXX/V4XX,XX,XXX,XXX,XXXX/V5XX,XX,XXX,XXX,XXXX/V6XX,XX,XXX,XXX,XXXX/V7040,083,00061,22222222222111/V8040,089,00061,22222222222111,/

TLeconte, 2018-09-10:
Yes this ACARS message must have been stored.
If it's not the case, it's a bug. Could yuo confirm that it had not be stored in db by acarsserv ?

ziorick, 2018-09-10:
Confirm. No message stored.
Could be interesting implement an http(s) method to send datas to a server?

acarslogger, 2018-09-12:
This is not an acars message. This is a VDL2 Ground Station broadcast message. Every VDL2 ground station periodically send a broadcast message addressed to 'all aircraft' giving the ground station location (lat/lon) and closest airport.

ziorick, 2018-09-13:
Can u save these messages on database?

TLeconte, 2018-09-22:
Just add -U -E and -G options for more configurable  json output

Hopes it's sufficient

closed 2018-09-22 by TLeconte