TLeconte/vdlm2dec #27: JSON -j UDP packet truncates at 500 bytes

acarslogger, 2018-09-24:
Hi Thierry,

The JSON UDP -j output is currently truncating UDP packets at 500 bytes.
I'm not an expert in using 'c' , but found in 'out.c' there was a line which reads
**char pkt[500]**;.
So I changed my copy to 1000, recompiled and it worked -- until I got a UDP packet that was greater than 1000 bytes!!
The message had an acars message of 985 characters; when you add the preceding data in JSON format it broke the 1000 barrier.
I know in traditional (analogue/2400bps) acars messages the maximum length of the acars message is 220 bytes. Is there a corresponding maximum length (1024?) defined for VDL2 processing?

Regards
Stuart

TLeconte, 2018-09-24:
Fixed a lazy programmer bug.
Thanks for the report
Yes there is a max around 16K !
Now with -U in json , man could get big packet (by the way UDP is now not really a good idea).

PS : You seems to receive more "interesting" traffic than me.

closed 2018-09-24 by TLeconte

acarslogger, 2018-09-25:
Hi Thierry,

Thanks for the corrected version of vdlm2dec.

If you are interested here is 1 example of an acars message greater than 1000 bytes.
The first part of the message (Msg F21A) is 998 bytes long.
The continuation / second part of the message (Msg F21B) is the remaining 340+ bytes.
If an acars message can be up to 16k, I just wonder whether there is a cut off set in the aircraft avionics which cuts a long message at 1000 or 1024 bytes.

Stuart

[#5 (F:136.975 P:-5.6) 21/09/2018 10:33:43.941 --------------------------------
Command from Aircraft:344446 (airborne) to GroundD:2651D7 
Frame-I: Ns:6 Nr:2
ACARS
Aircraft reg: EC-LUT Flight id: UX0238
Mode: 2 Msg. label: H1
Block id: 2 Ack: !
Msg. no: F21A
Message :
#M1BPOSN51173W000382,RODNI,103340,227,SITET,104320,ETRAT,M38,269063,164/FHSITET,104320,259K,3616,149,M48,266101,459,N,161,78.ETRAT,104639,255K,3700,146,M48,265103,460,N,166,26.DVL,104938,255K,3700,143,M48,266110,460,N,165,23.LGL,105345,255K,3700,140,M48,268120,460,N,165,32.SORAP,105552,255K,3700,138,M48,269117,460,N,166,17.BENAR,105758,255K,3700,136,M48,271113,460,N,165,17.VANAD,110112,255K,3700,133,M48,273107,460,N,165,26.AMB,110421,255K,3700,130,M48,276100,460,N,165,25.BALAN,111126,255K,3700,124,M48,283083,460,N,181,55.EVPOK,111916,255K,3700,118,M48,289065,460,N,167,64.NARAK,112805,255K,3700,110,M48,305051,460,N,167,73.GAI,113035,255K,3700,108,M48,312048,460,N,171,21.LOMRA,113502,255K,3700,104,M48,323045,460,N,176,37.ROCAN,113953,255K,3700,100,M48,337044,460,N,176,40.IBRAP,114211,255K,3700,98,M48,343043,460,N,165,19.LUKEV,114635,255K,3700,94,M48,354042,460,N,165,37.LORES,115704,279K,1906,91,M23,313013,367,Y,166,78.POSBA,120017,250K,1404,91,M13,319011,306,Y,151,19.POS,120416,240K,45

[#5 (F:136.975 P:-5.7) 21/09/2018 10:33:46.266 --------------------------------
Command from Aircraft:344446 (airborne) to GroundD:2651D7 
Frame-I: Ns:7 Nr:3
ACARS
Aircraft reg: EC-LUT Flight id: UX0238
Mode: 2 Msg. label: H1
Block id: 3 Ack: !
Msg. no: F21B
Message :
#M1B0,90,M2,325009,272,Y,151,20.BAVUM,120711,224K,450,89,P4,325005,240,Y,185,13.OBUMU,120908,180K,250,88,P8,325003,189,Y,239,7.FF24L,121006,160K,250,88,P10,325002,166,Y,239,3.RW24L,121252,160K,6,85,P15,000000,160,Y,239,8.420,,,42,,,000000,,Y,239,0.INTC,,185K,,,,000000,,Y,164,3.VECTOR,,,300,,,000000,,Y,193.LEPA,,150K,3,,,000000,,N,258,1/XX0CBD

TLeconte, 2018-09-25:
No  it's a vdl mode 2 message that could be 16K long.
I did not expect  that ACARS message send by vdlm2 could be longer than "classical" ACARS.
I have to look at my code about that.

acarslogger, 2018-09-25:
Acars messages on L-Band are also greater than the "classical" 220 bytes. An example follows.

Stuart

ACARS mode: 2  Aircraft reg: .OO-SFN (Jaero L-Band GES:43)
Message label: 30  Block id: L [Uplink]
Message content:-
LOADSHEET FINAL 1114 EDNO1
SN371/29      29JUN18
BRU DLA OOSFN    2/8
ZFW 165136 MAX 167000  L
TOF 48000
TOW 213136 MAX 215000  L
TIF 39900
LAW 173236 MAX 177000
UNDLD 1864
PAX/25/242 TTL 278
DOI      96.7
LIZFW   115.9
LITOW   120.2
MACZFW   28.3
MACTOW   28.3
FUEL DENSITY           0.8
A25 B45 C93 D104
SEATROW TRIM
SI A330-300 BW
DOW 124804
SALEABLE CONFIGURATION 30C/256Y
PAX WEIGHTS USED M94 F76 C35 I0
PANTRY CODE  S
CHECKED BAGGAGE PIECES DLA 2/Y/185
DLA 3/Y/109 4/C/23/Y/7 5/Y/20
NSI 2/Y/72 3/Y/112 4/C/25/Y/24 5/Y/35
LOAD IN CPTS 0/0 1/0 2/8577 3/7612 4/
2110 5/1537
DLA C    1993 M     130 B   344/   7356
O     110 T       0
NSI C    3034 M       0 B   268/   5674
O       0 T       0
NOTOC: YES
ICE/1/52
LIZFW LIMITS +   77.3/+  171.9
NOW: +  115.9
LITOW LIMITS +   43.2/+  161.6
NOW: +  120.2
PREPARED BY PIETER/DECLERCK  2 7883191
ISSUANCE DATE AND TIME: 29JUN 09:14 UTC
----------------------------------------------------------[29-06-18 09:15:39]