TLeconte/vdlm2dec #29: Program aborts after Updating 78b23c5..7dc3a78

DickvanNoort, 2018-09-24:
Hi Thierry,
This evening I upgraded using git pull which showed:
   78b23c5..7dc3a78  master     -> origin/master
 * [new tag]         v2.1       -> v2.1
Updating 78b23c5..7dc3a78

I added the -G option which makes the command that is run from a .sh script looks like this:
/home/pi/vdlm2dec/vdlm2dec -v -G -j 176.9.123.123:60003 -l "/home/pi/vdlm2dec/logs/vdl2_rtlsdr-11_$(date +"%Y%m%d").log" -p -1 -r 00000011 136.725 136.775 136.825 136.875 136.975 >> "/home/pi/vdlm2dec/logs/process-vdlm2dec.log" 2>&1 & 

The .sh script is coded so that it will restart the program when it fails.
The last few hours it has failed and restarted 6 times..
The previous version ran 24/7 without any problems.

I can't find anything in any of the logfiles.

Let me know if I can be of any help.

Cheers,
Dick

TLeconte, 2018-09-25:
Do you have the same problem without -G ?
I don't receive any ground station at home, so it's difficult to test.

DickvanNoort, 2018-09-25:
I will try without the –G, but I don’t  receive any ground stations here.
So if it is related to the –G it is not because ground messages are received..

Just started without –G.
Lets wait a few hours.

 

Cheers,

Dick

DickvanNoort, 2018-09-25:
Same problem without the –G.

Program aborted after 10 minutes and was restarted.

 

Cheers,

Dick

dewoert, 2018-09-25:
Same problem here, perhaps this error-message points in a good direction:
`./vdl.sh: line 8: 14395 **Segmentation fault**      /home/pi/vdlm2dec/./vdlm2dec -j 176.9.123.123:60003 -g 490 -p 50 -r 0 -G 136.725 136.775 136.875 136.975`

DickvanNoort, 2018-09-26:
Easy reproducable here by using this with just  -v and -l :
 /home/pi/vdlm2dec/vdlm2dec -v -l "/home/pi/vdlm2dec/logs/vdl2_rtlsdr-11_$(date +"%Y%m%d").log" -p -1 -r 00000011 136.725 136.775 136.825 136.875 136.975 >> "/home/pi/vdlm2dec/logs/process-vdlm2dec.log" 2>&1

Segmentation fault

DickvanNoort, 2018-09-26:
Also this very basic one fails

/home/pi/vdlm2dec/vdlm2dec  -r 1 136.725
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000011

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain : 49.600000
Set center freq. to 136775000Hz
Exact sample rate is: 2000000.052982 Hz

[#1 (F:136.725 P:-2.6) 26/09/2018 09:35:05.722 --------------------------------
Command from Aircraft:440128 (airborne) to GroundD:11982A
Frame-I: Ns:1 Nr:1
ACARS
Aircraft reg: OE-IJZ Flight id: U287NG
Mode: 2 Msg. label: _d
Block id: 2 Ack: U
Msg. no: S55A
Segmentation fault

TLeconte, 2018-09-26:
Strange, I don't have any problems ...
Could you ( yes  I ask a lot ) : 

In Makefile.rtl change in 1st line -Ofast by -g

_make -f Makefile.rtl clean
make -f Makefile.rtl_ 

_ulimit -c unlimited_

run your script until you have a lest one fail
You must find in the current directory a core.xxx file

install gdb and run 

gdb vdlm2dec -c core.xxxx

type : _where_ 

send me the result

DickvanNoort, 2018-09-26:
Hi Thierry,

 

No you do not ask a lot.
Happy to help if I can..

 

This is what I did:
Changed the Makefile.rtl

 

pi@MYPI-16B:~/vdlm2dec $ make -f Makefile.rtl clean

pi@MYPI-16B:~/vdlm2dec $ make -f Makefile.rtl

gcc -g -Wall -pthread -D WITH_RTL  -I /usr/local/include/librtlsdr   -c -o main.o main.c

gcc -g -Wall -pthread -D WITH_RTL  -I /usr/local/include/librtlsdr   -c -o rtl.o rtl.c

gcc -g -Wall -pthread -D WITH_RTL  -I /usr/local/include/librtlsdr   -c -o d8psk.o d8psk.c

gcc -g -Wall -pthread -D WITH_RTL  -I /usr/local/include/librtlsdr   -c -o vdlm2.o vdlm2.c

gcc -g -Wall -pthread -D WITH_RTL  -I /usr/local/include/librtlsdr   -c -o viterbi.o viterbi.c

gcc -g -Wall -pthread -D WITH_RTL  -I /usr/local/include/librtlsdr   -c -o rs.o rs.c

gcc -g -Wall -pthread -D WITH_RTL  -I /usr/local/include/librtlsdr   -c -o crc.o crc.c

gcc -g -Wall -pthread -D WITH_RTL  -I /usr/local/include/librtlsdr   -c -o out.o out.c

gcc -g -Wall -pthread -D WITH_RTL  -I /usr/local/include/librtlsdr   -c -o outacars.o outacars.c

gcc -g -Wall -pthread -D WITH_RTL  -I /usr/local/include/librtlsdr   -c -o label.o label.c

gcc -g -Wall -pthread -D WITH_RTL  -I /usr/local/include/librtlsdr   -c -o outxid.o outxid.c

gcc -g -Wall -pthread -D WITH_RTL  -I /usr/local/include/librtlsdr   -c -o cJSON.o cJSON.c

gcc  main.o rtl.o d8psk.o vdlm2.o viterbi.o rs.o crc.o out.o outacars.o label.o outxid.o cJSON.o -o vdlm2dec -lm -pthread  -lusb-1.0  -L/usr/local/lib -lrtlsdr -lrt

pi@MYPI-16B:~/vdlm2dec $ ulimit -c unlimited

 

pi@MYPI-16B:~/vdlm2dec $ ./vdlm2dec  -r 1 136.725

Found 1 device(s):

  0:  Realtek, RTL2838UHIDIR, SN: 00000011

 

Using device 0: Generic RTL2832U OEM

Found Rafael Micro R820T tuner

Tuner gain : 49.600000

Set center freq. to 136775000Hz

Exact sample rate is: 2000000.052982 Hz

 

[#1 (F:136.725 P:-4.3) 26/09/2018 18:09:36.316 --------------------------------

Command from Aircraft:4AC954 (airborne) to GroundD:11900A

Frame-I: Ns:1 Nr:2

Segmentation fault (core dumped)

 

And here is the result:

 

pi@MYPI-16B:~/vdlm2dec $ gdb vdlm2dec -c core

GNU gdb (Raspbian 7.12-6) 7.12.0.20161007-git

Copyright (C) 2016 Free Software Foundation, Inc.

License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>

This is free software: you are free to change and redistribute it.

There is NO WARRANTY, to the extent permitted by law.  Type "show copying"

and "show warranty" for details.

This GDB was configured as "arm-linux-gnueabihf".

Type "show configuration" for configuration details.

For bug reporting instructions, please see:

<http://www.gnu.org/software/gdb/bugs/>.

Find the GDB manual and other documentation resources online at:

<http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".

Type "apropos word" to search for commands related to "word"...

Reading symbols from vdlm2dec...done.

[New LWP 4647]

[New LWP 4638]

[New LWP 4645]

[New LWP 4646]

[Thread debugging using libthread_db enabled]

Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Core was generated by `./vdlm2dec -r 1 136.725'.

Program terminated with signal SIGSEGV, Segmentation fault.

#0  0x0001a438 in cJSON_Delete (item=0xc125eb88) at cJSON.c:220

220             next = item->next;

[Current thread is 1 (Thread 0x75bff460 (LWP 4647))]

(gdb) where

#0  0x0001a438 in cJSON_Delete (item=0xc125eb88) at cJSON.c:220

#1  0x000169b4 in out (blk=0x75c045f0,

    hdata=0x75bfaee8 "~\024b\004PP\324H+B\033\377\201\334\025~\255+\036\312ʣ1\253\327\312\377Y\n\265\357\356~^\022k\211-+\034\375e\030\375V\334\336J\033\266\235\256\352<6b\031ƌ7\302\305\370\005\376ӘF\360\350\336\263\004\322Y\347ɕkT\300\201j\227\222a2\360\206\225j\242\t\241\211\066rK\206\367[\251\244~^\344\024\340\277I\240\006R\231\315\r\vrO\024\001\177*I\276N\245\202\061\250_5\260\207\331\363.\210\037n\361\350^\202e\215\027\241b%\302\207\272\f\177\034=\377\317_KM\373!L\232\363\307\001m\344\317\301\242\a˯\253\030X\016\237\201\035\350\261ݟ\312?\363wˠwym\215\227", <incomplete sequence \372>..., l=16) at out.c:345

#2  0x0001416c in check_frame (blk=0x75c045f0,

    hdata=0x75bfaee8 "~\024b\004PP\324H+B\033\377\201\334\025~\255+\036\312ʣ1\253\327\312\377Y\n\265\357\356~^\022k\211-+\034\375e\030\375V\334\336J\033\266\235\256\352<6b\031ƌ7\302\305\370\005\376ӘF\360\350\336\263\004\322Y\347ɕkT\300\201j\227\222a2\360\206\225j\242\t\241\211\066rK\206\367[\251\244~^\344\024\340\277I\240\006R\231\315\r\vrO\024\001\177*I\276N\245\202\061\250_5\260\207\331\363.\210\037n\361\350^\202e\215\027\241b%\302\207\272\f\177\034=\377\317_KM\373!L\232\363\307\001m\344\317\301\242\a˯\253\030X\016\237\201\035\350\261ݟ\312?\363wˠwym\215\227", <incomplete sequence \372>..., l=16) at vdlm2.c:59

#3  0x00014548 in blk_thread (arg=0x0) at vdlm2.c:141

#4  0x76ea5fc4 in start_thread (arg=0x75bff460) at pthread_create.c:335

Backtrace stopped: Cannot access memory at address 0x43059572

 

 

Hope this helps..

 

Cheers,

Dick

TLeconte, 2018-09-26:
I fear that there is more than one problem ...
Your trace shows  a real problem without -J or -j. I hope that it is solved with my last out.c commit.
Could you check that ?
But It does not explain problems with -j or -J 
So please, redo tests with and without -J/-j and give me the gdb result if there is a segfault (be sure to remove core.xx before)

DickvanNoort, 2018-09-26:
> I fear that there is more than one problem ...
> Your trace shows a real problem without -J or -j. I hope that it is solved with my last out.c commit.
> Could you check that ?

Trying to update fails with these for me unknown messages:
pi@MYPI-16B:~/vdlm2dec $ git pull

*** Please tell me who you are.

Run

  git config --global user.email "you@example.com"
  git config --global user.name "Your Name"

to set your account's default identity.
Omit --global to set the identity only in this repository.

fatal: empty ident name (for <(null)>) not allowed

I suppose I first have to update to the latest commit before I do additional tests.
Correct??

> So please, redo tests with and without -J/-j and give me the gdb result if there is a segfault (be sure to remove core.xx before)
Do you mean with and without -J/-j  but always without -l  ??

TLeconte, 2018-09-26:
You must have a git config problem ???

Yes update to last commit before retesting without -J / -j
then if it works try with -J/-j

DickvanNoort, 2018-09-27:
>  You must have a git config problem ???
Yes that is what it looks like, but just like that??
Or is this because I changed the Makefile.rtl >>
And more importantly, how do I fix this.
I can't remember making doing any git config stuff on my RPi.

Anyway I probably have no time for this today.

PS
When I run it with -l only it fails with the first message.
When I run it normally it runs much longer and does create a logfile with valid records.
Is it possible that one bug influences the other?

DickvanNoort, 2018-09-27:
Not sure why I could not do a git pull like I did many times before.
So I ran the two commands:
git config --global user.email "you@example.com"
git config --global user.name "Your Name"

And was able to do a git pull which resulted in this:
Merge made by the 'recursive' strategy.
 out.c | 3 ++-
 1 file changed, 2 insertions(+), 1 deletion(-)

So now I am on Latest commit 5249a2a .
Compiled and ran: ./vdlm2dec  -r 1 136.725   for a few minutes without problems.
Now rebooted to start the normal operation which uses this command:
/home/pi/vdlm2dec/vdlm2dec -v -j 176.9.123.123:60003 -l "/home/pi/vdlm2dec/logs/vdl2_rtlsdr-11_$(date +"%Y%m%d").log" -p -1 -r 00000011 136.725 136.775 136.825 136.875 136.975 >> "/home/pi/vdlm2dec/logs/process-vdlm2dec.log" 2>&1 &  	

So far looking good..
 
Thanks.

acarslogger, 2018-09-27:
Hi Thierry,

I've run a clean install of the latest version (with fix unintialized global) with various combinations of -l -J and -j and all looks good so far. No crashes.
Can you confirm that the -G -E and -U only affect the -j and -J output. The basic -l log file (without the -J) is not affected.

Thanks
Stuart

TLeconte, 2018-09-27:
No -E -G and -U affect -l and -j/-J output.

-E and -G affect -l and -j/-J output in the same manner (without these options, empty (resp. from ground) packets are not outputted.

For -U it's less than perfect: without -U and with -j/-J undecoded packet are not outputted at all , 
without-U and with -l the packet are outputted but there content could not be displayed. It was not so easy to do better due to the structure of the code.

acarslogger, 2018-09-27:
Thanks Thierry.

vdlm2dec still running OK with no crashes.

I'll try various combinations of the -G -E and -U command line options.

Thanks again
Stuart