TLeconte/vdlm2dec #30: Support for AirSpy MINI

istamov, 2018-09-25:
Hello,

First of all - thank you for your efforts in developing this great software.
Is it possible to also add support for AirSpy MINI - it differs from the regular AirSpy R2 in terms of the supported sample rates:	

```
### Regular AirSpy R2
Board ID Number: 0 (AIRSPY)
Firmware Version: AirSpy NOS v1.0.0-rc10-3-g7120e77 2018-04-28
Supported sample rates:
	10.000000 MSPS
	2.500000 MSPS
```
```
### AirSpy MINI
Board ID Number: 0 (AIRSPY)
Firmware Version: AirSpy MINI v1.0.0-rc10-3-g7120e77 2018-04-28
Supported sample rates:
	6.000000 MSPS
	3.000000 MSPS
```

I was able to compile and run vdlm2dec by changing SDRINRATE in file "vdlm2.h" from "5000000" to "6000000", but not sure this is the only change that is needed.

```
#ifdef WITH_AIR
#define SDRINRATE 6000000
#define SDRCLK  1250
#endif
```

Thanks in advance!

TLeconte, 2018-09-25:
No I will not be sufficient.
I will try to think about a solution, but  I will not be able to test it, and :

>  no tested, does not work

istamov, 2018-09-25:
No problem if you need any testing for the new code or even remote access to an RPi3 with the AirSpy MINI attached.
But if it will be too much of a rework, I will just try to get an AirSpy R2 to replace the MINI.

TLeconte, 2018-09-25:
Try : 
#define SDRINRATE 3000000
#define SDRCLK  750

and in function chooseFc (air.c) comment the 2 lines with _airspy_r820t_write_

istamov, 2018-09-25:
Tried with the following changes:

In file "vdlm2.h":
```
#ifdef WITH_AIR
#define SDRINRATE 3000000
#define SDRCLK 750
#endif
```
In file "air.c":
```
static unsigned int chooseFc(unsigned int minF,unsigned int maxF)
{
        unsigned int bw=maxF-minF+2*STEPRATE;
        unsigned int off;
        int i,j;

        for(i=7;i>=0;i--)
                if((r820t_hf[5]-r820t_lf[i])>=bw) break;
        if(i<0) return 0;

        for(j=5;j>=0;j--)
                if((r820t_hf[j]-r820t_lf[i])<=bw) break;
        j++;

        off=(r820t_hf[j]+r820t_lf[i])/2-SDRINRATE/4;

        //airspy_r820t_write(device, 10, 0xB0 | (15-j));
        //airspy_r820t_write(device, 11, 0xE0 | (15-i));

        return((maxF+minF)/2+off);
}
```

But the result is as before:
```
$ ./vdlm2dec -v -U -E -G 136.875 136.975
did not find needed sampling rate
Unable to init input
```

TLeconte, 2018-09-26:
My bad.
try with 
#ifdef WITH_AIR
#define SDRINRATE 6000000
#define SDRCLK 1500
#endif

and keep the commants

istamov, 2018-09-26:
Thank you, that seems to do the trick (changing the defines and commenting the "airspy_r820t_write" lines), now I receive messages:

```
$ ./vdlm2dec -v -U -E -G 136.725 136.775 136.825 136.875 136.975
Set freq. to 137223862 hz

[#5 (F:136.975 P:1.7) 26/09/2018 17:59:52.472 --------------------------------
Response from Aircraft:4B9061 (airborne) to GroundD:29BCD8 
Frame-S: Nr:6 RR

[#5 (F:136.975 P:3.0) 26/09/2018 17:59:55.687 --------------------------------
Command from Aircraft:406D8E (airborne) to GroundD:1091AA 
Frame-I: Ns:0 Nr:1
unknown data
1b ff ca 1e 28 f6 40 00 f7 85 00 1e 04 00 00 00    |....(.@.........|
04 00 00 00 02 03 04 bc 5c d9 31 b4 da 00 39 3b    |........\.1...9;|
90 20 9f 2e d8 48 08                               |. ...H..........|

[#5 (F:136.975 P:3.1) 26/09/2018 17:59:56.502 --------------------------------
Response from Aircraft:406D8E (airborne) to GroundD:1091AA 
Frame-S: Nr:2 RR
```

istamov, 2018-10-08:
The MINI is on a remote site currently connected to a wideband Diamond D-130NJ discone antenna, but I guess it will still receive some ADS-B signals.
If you are not in a hurry, I can visit the place in about a week and attach an ADS-B optimised antenna (FlightAware 5.5 dBi) to it, so we can test how it performs.

TLeconte, 2018-10-08:
OK, that's great.
No hurry at all. I have to do the mods before, but I wanted to be sure that someone could test them.

TLeconte, 2018-10-09:
OK I just have committed changes for airspy mini.
When you will get the time : 
- Get the last version
- add -D AIRSPY_MINI to the CFLAGS in Makefile
- make clean
- make

GIve your results in Issue of adsbdec

Thanks

istamov, 2018-10-10:
Thanks, I have tried the latest git (by the way it shows the last modification as 3 days ago?), but I still get "did not find needed sampling rate".

I have reported the full command output in the issue opened by @papasven, where he has also provided more details.