TLeconte/vdlm2dec #32: Program aborts with "75 Read async -1"

DickvanNoort, 2018-09-28:
Installed the Latest commit 45c3248 from scratch 
and using these options
/home/pi/vdlm2dec/vdlm2dec -v -j 176.9.123.123:60003 -l "/home/pi/vdlm2dec/logs/vdl2_rtlsdr-11_$(date +"%Y%m%d").log" -p -1 -r 00000011 136.725 136.775 136.825 136.875 136.975
Found 1 device(s):
  0:  Realtek, RTL2838UHIDIR, SN: 00000011

Using device 0: Generic RTL2832U OEM
Found Rafael Micro R820T tuner
Tuner gain : 49.600000
Set center freq. to 137025000Hz
Exact sample rate is: 2000000.052982 Hz

the program aborts with errors like these:
75
Read async -1

pi@MYPI-16B:~/vdlm2dec $ vdlm2dec: no process found
75
Read async -1

pi@MYPI-16B:~/vdlm2dec $ 100
Read async -1

DickvanNoort, 2018-09-29:
After rebooting the RPi all is good and back to normal.
No idea what went wrong and why..

Thanks,
DIck

closed 2018-09-29 by DickvanNoort

TLeconte, 2018-09-29:
Probably another vdlm2dec process was running in the background (or any process that use the rtl-sdr).