TLeconte/vdlm2dec #36: High CPU Use

acarslogger, 2018-12-13:
Hi Thierry,

I'm running the latest commit cf6d0724571dad9b505832903983b78c1d9e009c with the new -b and libacars.
Previous versions have used about 45% of cpu on my Raspberry Pi v3.
When using 'top', this latest version appears to be running at 195% cpu; although I assume it is spread across all cpu's.
Why is the CPU use so high?
Regards
Stuart

TLeconte, 2018-12-14:
Strange. I did not change the signal processing code which is the part that consume CPU.
I think it's more a Makefile vs CMake  problem.
I naively hoped that CMake will use the good compile options depending of target.

Could you try to add to CMakeLists.txt the line :
add_compile_options(-O3 -mcpu=cortex-a53  -mfpu=neon-fp-armv8)
just after add_executable.

Then 
rm -rf build
and then redo all the compile procedure (make build  etc ...)

I do that blindly as I don't have a RP3

TLeconte, 2018-12-15:
Not normal; 
What compile options did you use in the Makefile for the RP3 ?