TLeconte/vdlm2dec #39: IQ values in rtl.c

BradWalker, 2019-06-04:
Inside of in_callback() the IQ values get adjusted by 127.37..

How did you determine this is a correct value to adjust them?

Thanks.

```
static void in_callback(unsigned char *rtlinbuff, uint32_t nread, void *ctx)
{
..
	for (i = 0; i < RTLINBUFSZ;) {
		float r, g;
		r = (float)rtlinbuff[i] - **(float)127.37;**
		i++;
		g = (float)rtlinbuff[i] - **(float)127.37;**
		i++;
		Cbuff[i / 2] = r + g * I;
	}
..
}
```

TLeconte, 2019-06-06:
If you average the value of IQ over a long time on a dummy load, you'll find a value a little less than 127.5.
It could depend of each device and temperature. 
I don't remember why I choose .37, perhaps it the result of a measurement or a value commonly used in other codes (perhaps from osmosdr )
It's an  "experimentaly good value" :-)

BradWalker, 2019-06-13:
Thanks! Makes sense..

closed 2019-06-13 by BradWalker