TLeconte/vdlm2dec #41: json/UDP feed stops working randomly

marcus-aa, 2020-01-06:
Sometimes the udp feed just stops working, until the process is killed and re-started. It can work for 30 days and suddenly stop. Local decoding and logging to text file continues normally.

As there is no state of a UDP connection that can be monitored it is hard to detect and restart.

command line is 

`/usr/local/bin/vdlm2dec -l /var/log/vdlmlog.txt -g 430 -p 60 -i P-ESSN9-V -j 176.9.123.123 60003 -G -r 1 136.725 136.775 136.825 136.875 136.975`

TLeconte, 2020-01-06:
You have to restart vdlm2dec or acarsserv ? 
I could try to see if a better error logging will help.
Could be an external  network reset that breaks vdlm2dec/acarsserv send/receive

marcus-aa, 2020-01-06:
If I restart vdlm2dec it will start working again, but this is a headless, remote client and I need to keep it running unattended. 

acarsserv does not need to be restarted, also I don't control it so I can't test if a restart will re-enable the client. But as other clients keep working against acarsserv it seems unlikely.

TLeconte, 2023-02-05:
Try with v2.2 , it adds auto-reconnect

closed 2023-02-05 by TLeconte