TLeconte/vdlm2dec #42: Issue building on Alpine Linux amd64

kevinelliott, 2020-04-29:
It looks like there's some kind of header issue with `timeval` on Alpine Linux.

```
#34 3.586 [ 15%] Building C object CMakeFiles/vdlm2dec.dir/crc.c.o
#34 3.660 [ 23%] Building C object CMakeFiles/vdlm2dec.dir/d8psk.c.o
#34 4.231 [ 30%] Building C object CMakeFiles/vdlm2dec.dir/label.c.o
#34 4.255 In file included from /vdlm2dec/label.c:5:
#34 4.255 /vdlm2dec/vdlm2.h:47:17: error: field 'tv' has incomplete type
#34 4.255   struct timeval tv;
#34 4.255                  ^~
#34 4.258 In file included from /vdlm2dec/label.c:6:
#34 4.258 /vdlm2dec/acars.h:47:24: error: field 'ts' has incomplete type
#34 4.258          struct timeval ts,tl;
#34 4.258                         ^~
#34 4.258 /vdlm2dec/acars.h:47:27: error: field 'tl' has incomplete type
#34 4.258          struct timeval ts,tl;
#34 4.258                            ^~
#34 4.268 make[2]: *** [CMakeFiles/vdlm2dec.dir/build.make:102: CMakeFiles/vdlm2dec.dir/label.c.o] Error 1
#34 4.270 make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/vdlm2dec.dir/all] Error 2
#34 4.270 make: *** [Makefile:130: all] Error 2
```

TLeconte, 2023-03-05:
Could you check with V2.3

kevinelliott, 2023-03-08:
> Could you check with V2.3

I'll give it a shot and let you know!