TLeconte/vdlm2dec #43: multiple UDP outputs?

kidshare, 2020-05-06:
Is it possible to output multiple UDP such as -j 192.168.1.2:8880 -j feed.acars.io:5550

TanerH, 2020-05-25:
@kidshare if you're looking for a quick work-around, you can check out 
Samplicator - https://github.com/sleinen/samplicator

It will run a daemon on your local machine that will listen on a port and then replicate the UDP packets to multiple destinations.  You can simply point vdlm2dec to localhost:port

kevinelliott, 2020-09-05:
@kidshare I second what @TanerH said here. Several of us are using this solution quite well (though native would be a great addition too).

kevinelliott, 2020-09-16:
@kidshare It's not complete yet (read: alpha) and not for the faint at heart... but I've just published https://github.com/airframesio/airframes-client which will wrap `vdlm2dec` as well as `acarsdec` and `dumpvdl2` and let you set an unlimited number of destination outputs (log files, udp remotes, and yes, also airframes.io / acars.io).