TLeconte/vdlm2dec #44: cmake not finding libacars-2

TanerH, 2020-05-25:
The CMakeLists.txt is only looking for libacars, and not libacars-2 (which is the latest from https://github.com/szpajder/libacars )

This diff will check for acars-2, first, and then acars, before giving up on that library.

(Forgive the indenting, please adjust as you'd like)

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index c619ab7..b86ea58 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -5,13 +5,20 @@ add_compile_options(-Ofast -march=native )

 add_executable(vdlm2dec cJSON.c  crc.c  d8psk.c  label.c  main.c  outacars.c  out.c  outxid.c  rs.c  vdlm2.c  viterbi.c )

-find_library(LIBACARS acars)
-if(LIBACARS)
-message ( STATUS "Using libacars")
-add_definitions(-DHAVE_LIBACARS )
-target_link_libraries( vdlm2dec acars)
+find_library(LIBACARS2 acars-2)
+if(LIBACARS2)
+       message ( STATUS "Using libacars-2")
+       add_definitions(-DHAVE_LIBACARS )
+       target_link_libraries( vdlm2dec acars-2)
 else()
-message ( STATUS "Not using libacars")
+       find_library(LIBACARS acars)
+       if(LIBACARS)
+               message ( STATUS "Using libacars")
+               add_definitions(-DHAVE_LIBACARS )
+               target_link_libraries( vdlm2dec acars)
+       else()
+               message ( STATUS "Not using libacars (v1 or v2)")
+       endif()
 endif()

 option(rtl "Compiling for rtl sdr" )
```

kevinelliott, 2020-09-05:
@TanerH Want to put this in a PR?

kevinelliott, 2020-09-16:
@TanerH Ping... if you end up having a moment... or not if you don't think it's useful anymore :)

cdschuett, 2020-12-23:
The change suggested by @TanerH fixed the issue with cmake not finding the libacars library, but I got compile errors when the script was compiling outacars. It couldn't find the header files for the library.
I fixed it by adding:
`if(LIBACARS2)
       message ( STATUS "Using libacars-2")
       add_definitions(-DHAVE_LIBACARS )
       target_link_libraries( vdlm2dec acars-2)
       target_include_directories( vdlm2dec PUBLIC "/usr/local/include/libacars-2")
 else()
       find_library(LIBACARS acars)
       if(LIBACARS)
               message ( STATUS "Using libacars")
               add_definitions(-DHAVE_LIBACARS )
               target_link_libraries( vdlm2dec acars)
               target_include_directories( vdlm2dec PUBLIC "/usr/local/include/libacars")`

There might be a better way to do this, but this worked for me.